MatMoul/g810-led #229: Weird behaviour on g810, ubuntu 20.04

hugwi, 2020-06-05:
I'm currently on a ubuntu 20, installed lib-1.0.0 and g810-led. However g810 is behaving a 
bit unexpected. 

Running any of the sample commands outputs 

```
$ g810-led -k logo ff0000
Matching or compatible device not found !
```
But the logitech keyboard is supported and found.
```
$ g810-led --list-keyboards
Device: Logitech - Gaming Keyboard G810
	Vendor ID: 046d
	Product ID: c337
	Serial Number: 087536763934
```
which is the same information retrieved by lsusb
`Bus 001 Device 005: ID 046d:c337 Logitech, Inc. Gaming Keyboard G810`

When instead installing manually i get this output instead.

`1]    125628 segmentation fault (core dumped)  g810-led --list-keyboards`

Any suggestions on what can be the problem here?

viniciusbo, 2020-06-07:
That's happening to me as well. I've been using g810-led for a couple days, and today I just stopped working with that same message "Matching or compatible device not found !". I've installed g810-led using Portage on Gentoo.

viniciusbo, 2020-06-07:
Running g810-led commands using superuser (su or sudo) works fine.

MatMoul, 2020-08-11:
sudo or udev rules...