MatMoul/g810-led #230: Support for G512 SE

Andre0711er, 2020-06-05:
Hi i have a brand new G512 SE here, which is not detected as G512 (yeah, it's special Edition).

Kernel give me following Info:

[28978.000443] usb 1-10: new full-speed USB device number 8 using xhci_hcd
[28978.149537] usb 1-10: New USB device found, idVendor=046d, idProduct=c342, bcdDevice=50.01
[28978.149541] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[28978.149544] usb 1-10: Product: G512 SE
[28978.149546] usb 1-10: Manufacturer: Logitech
[28978.149547] usb 1-10: SerialNumber: 0**********
[28978.151985] input: Logitech G512 SE as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:046D:C342.0007/input/input25
[28978.208623] hid-generic 0003:046D:C342.0007: input,hidraw0: USB HID v1.11 Keyboard [Logitech G512 SE] on usb-0000:00:14.0-10/input0
[28978.209358] input: Logitech G512 SE Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:046D:C342.0008/input/input26
[28978.272538] input: Logitech G512 SE Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:046D:C342.0008/input/input27
[28978.272619] hid-generic 0003:046D:C342.0008: input,hiddev2,hidraw1: USB HID v1.11 Keyboard [Logitech G512 SE] on usb-0000:00:14.0-10/input1

---

g512-led --list-keyboards shows me:

Device: Logitech - G512 SE
	Vendor ID: 046d
	Product ID: c342
	Serial Number: 0**********

g512-led --print-device give me:

Matching or compatible device not found !

Can you help?

kittencoder, 2020-06-09:
Owning a G512 Tactile and using the library version of this project I already noticed that my G512 is being identified as a G513. (not a problem for me, posting it in case that might help with the issue of identification problems).

Andre0711er, 2020-06-09:
for my side, it's solved. i need execute CMDs with sudo command. otherwise is not working.

MatMoul, 2020-08-11:
sudo , udev rules ....

mosulache, 2021-02-01:
for me, lsusb shows:
Bus 003 Device 007: ID 046d:c342 Logitech, Inc. G512 SE

to make it work i need to execute:

sudo g512-led -dv 046d -dp c342 -tuk 1 -a 0000FF
(to turn it all blue)

dreamhopping, 2021-04-08:
Launching it with ``sudo`` works fine for me!
[image]

PhilippeNoel1, 2021-11-03:
Works for me too with the `sudo g512-led -dv 046d -dp c342 -tuk 1 -a 0000FF` command