MatMoul/g810-led #233: G610 no longer supported?

rparenti, 2020-07-08:
$ g610-led --list-keyboards --print-device
Device: Logitech - Gaming Keyboard G610
	Vendor ID: 046d
	Product ID: c338
	Serial Number: 0D7235623434

$ g610-led -dv 046d -dp c338 -ds 0D7235623434 --print-device
Matching or compatible device not found !

rparenti, 2020-07-08:
No USB hub in use. Changing USB ports does not help.

rparenti, 2020-07-08:
Arch linux 5.7.7

$ lsusb | grep Logitech
Bus 002 Device 010: ID 046d:c338 Logitech, Inc. Gaming Keyboard G610

robfrancken, 2020-07-30:
I am having the same issue. My device id is different though:

```sh
Bus 001 Device 005: ID 046d:c333 Logitech, Inc. Gaming Keyboard G610
```

ceranco, 2020-09-06:
I have the exact same problem.
```
Bus 003 Device 003: ID 046d:c083 Logitech, Inc. G403 Prodigy Gaming Mouse
```

rparenti, 2020-10-30:
> Still no news about this ?

I believe I replied to this list stating that the problem went away when 
I reformatted the filename in /etc/udev/rules.d. The filename needs to 
start with a number on my system.

GaetanLepage, 2020-11-02:
You are right, it works !
Thank you

MatMoul, 2020-11-06:
$ g610-led, you're not root, use sudo or check if file /etc/udev/rules.d/g810-led.rules exist for use as user