MatMoul/g810-led #234: Numbers and slash do not change their colors

Matosinho, 2020-07-17:
First, congratulations for the project. I was really happy that I didn't needed to install windows just for that. I'm not able to change the colors of the number using `g512-led -k n9 ffffff`. I'm using g512. Also, the slash does not change its color either, even when I change the group colors. But, numbers and slash are changed when I use argument `-a`

msproesser, 2020-08-04:
Same here, using the g512 brazilian format ABNT I can't change the color of the slash key on ubuntu 20.04, even into ubuntu settings looks like the key doesn't exists instead there is a big shift key in place. As a workaround I changed the color using the fx color `g512-led -fx color all 0000ff` this way changes the whole keyboard color including the slash.

for the problem with the numbers probably you want to use `g512-led -k num9 ffffff`

dioni21, 2021-02-21:
People, please try my PR, https://github.com/MatMoul/g810-led/pull/241

The slash key on ABNT keyboards have an extra keycode, not yet supported.