MatMoul/g810-led #236: Matching or compatible device not found !

DavidFaria38, 2020-07-28:
Hello, i'm new to linux and ive been try to get into using linux as my daily driver for a while, im on Ubuntu 20.04.
I'm using a g213 and when i try to command to change or cycle through colors, for example:
```
    g213-led -a 00ff00 
```
it outputs:
```
     Matching or compatible device not found !
```

Thanks.

jonathanbossenger, 2020-07-31:
@DavidFaria38 I'm also on Ubuntu 20.04, I installed g810-led from the apt repo. Had the same problem as you, a reboot seems to have solved it, and I can now change colors etc.

MatMoul, 2020-08-11:
sudo ?

jonathanbossenger, 2020-08-12:
@MatMoul difficult to determine, as after a restart it works without needing to use sudo.

I'll take it home on the weekend and try it with my laptop, and see if sudo makes the difference.

pearsonk, 2020-08-12:
I've seen this question posed several times with similar resolution (being fine after reboot). One hypothesis I would have would be that even though udev rules are put in place, perhaps the postinstall script of any OS package isn't reloading/triggering the rules to have them take immediate effect. This would result in the behavior of permission denial until reboot causes the new rules to actually take effect.

j-applese3d, 2020-09-24:
`sudo` did the trick for me. So does un- and re-plugging in the keyboard.
Maybe the error message could suggest using `sudo` if user isn't running the command as root? Or to try unplugging and re-plugging in the keyboard?

As it is, the error makes the user think that their keyboard won't work with the software.

closed 2020-11-06 by MatMoul