MatMoul/g810-led #239: GPRO key lighting issue

wernerfletcher, 2020-08-07:
Good day,

I've been running this software for quite some time now.  I'm using the GPro Tenkeyless keyboard on Linux mint 19.2.  I have not made changes to any of the profiles, it's still the default since install of the software.

The issue is that the number 4 has a different colour to the other keys.  I've tried re-installing the software but the issue still persists.

Any idea why this could be happening?

Thank you.

MatMoul, 2020-08-11:
Hi, I've just bought a new key keyboard because one led on one key is ded on my last g810...
I suspect this is the same case.
Try g810-led -a ff0000
then g810-led -a 00ff00
then g810-led -a 0000ff
and saw if you show a differance on a key.

wernerfletcher, 2020-08-12:
I believe you are correct.  Red and green is perfect but the blue is a little dimmer than the rest of the keys.
Thank you for taking the time to help, much appreciated.

closed 2020-08-12 by wernerfletcher