MatMoul/g810-led #240: enhancing existing built-in profile

DFF-fred, 2020-09-10:
Hi!
Thank you very much first! Very happy to be able to use my keyboard under Linux and do all this.
Now the keyboard comes with a few built-in effect by pressing the FN+F5 keys. One of them is a black keyboard with some kind of propagating colourful wave propagating from the key pressed. Is there anyway to do that effect with g810-led and (hopefully) improved it?
Thank you.

MatMoul, 2020-09-14:
Hi,
Not sure what's your keyboard model but FN+XX is supported on g810...
But some FN+XX effect isn't supported from code.
You can check it with Windows...
Start the same effect from the Logitech Software and exit the Logitech Software.
If the effect is stopped, it's not possible to set it by code.

DFF-fred, 2020-09-15:
Oh I see. Didn't know that. I actually don't have Windows at all and therefore can't test. And my keyboard is a G512.
Thank you for the explanations.

closed 2020-11-06 by MatMoul