MatMoul/g810-led #242: unsure how to enable on start - ubuntu 20.04

daniejstriata, 2020-09-13:
Hi,

I'm not sure how to start the desired theme. I configured the g810-led-reboot  to use all_red but how do I get that to kick in on starting ubuntu 20.04?

Danie

daniejstriata, 2020-09-14:
I opted to run this cron:
`@reboot /usr/bin/g213-led -p /usr/share/doc/g810-led/examples/sample_profiles/all_red`

MatMoul, 2020-09-14:
Hi,
Replace the /etc/g810-led/profile file...
This is the way if you use udev (not sure how you have installed g810-led)

daniejstriata, 2020-09-15:
I installed it with apt.

I did edit it before. Mine looks like:
```
# Install your g810-led profile here
# (applied at boot or keyboard connection)
# See /usr/share/doc/g810-led/examples/sample_profiles
/usr/share/doc/g810-led/examples/sample_profiles/all_red
```

daniejstriata, 2020-09-17:
is the profile supposed to use the content of the profile?

MatMoul, 2020-11-06:
I'm not the maintainer of the ubuntu package but have you a /etc/g810-led/profile file?
If yes, you can edit or replace it with your profile (not the logitech pofile, the [gxxx-led profile](https://github.com/MatMoul/g810-led/blob/master/sample_profiles/keys_v_gradiant_fr_ch-latin1))