MatMoul/g810-led #243: G910 ... missing keys?

kdv666, 2020-09-19:
Perhaps I've missed something, but I don't seem to be able to influence the "M" keys, which are off, or the multimedia keys, which remain blue.  Sorry if it's clearly written somewhere and I've just not seen it ... if so, please point me in the right direction!

MatMoul, 2020-11-06:
Sorry, I don't have a g910 but looking in the code, g810-led have 2 args for m keys.
[-mr](https://github.com/MatMoul/g810-led/blob/5ee810a520f809e65048de8a8ce24bac0ce34490/src/main.cpp#L89) and [-mn](https://github.com/MatMoul/g810-led/blob/5ee810a520f809e65048de8a8ce24bac0ce34490/src/classes/Keyboard.cpp#L746) but it's undocumented...

kdv666, 2020-11-07:
Hi.  Many thanks for your reply.  I'm having no success, so still need suggestions.

g910-gkeys -mr 0 ( or -mr 1 ) returns without an error but doesn't seem to do anything.
g910-gkeys -mr xxxxxx ( for a colour ) returns an error
g910-gkeys -k mr xxxxxx returns an error
g910-gkeys -g multimedia xxxxxx does nothing

The source code seems to be written in C++, which I don't speak, so I really can't tell what I should be doing.

I know that at least the m-keys are able to be controlled, because the windows app works.

Thanks for your attention!

MatMoul, 2020-11-24:
I tag Help Vanted, maybe someone can be of help.

Corionis, 2021-02-01:
@MatMoul I too am a citizen of Sol 3, Milky Way :)

I have a Logitech Orion Spark USB keyboard with a couple of bad lights and a bad volume knob. The volume control caused me to replace it with a G910 Orion Spectrum.  However, that Spark keyboard is fine other than those annoyances.

I would be happy to donate that keyboard to your effort. That is, I will ship it to you, free. If you're OK with those limitations. 

For that matter I'd be happy drop-ship a new one :)

I am also working on a set of .bind files and scripts to change profiles and colors using the M-keys and refinements to start-up and clock scripts. Perhaps those could be added to your project after I get them working and your review?

I will also take a look at this particular issue when I have time. I'm an old C++ guy.

Thank you and the other contributors for solid software.