MatMoul/g810-led #247: Media Keys disabling

xdubx, 2020-10-07:
Hi,

have Pop OS 20.04. The problem is after the systemd services started the OS don't recognize the media keys any more and the volume slider on my Keyboard is the g815.

MatMoul, 2020-11-06:
g810-led is a one shot command and don't handle your keys...
How do you have installed it?

xdubx, 2020-11-07:
clone repo, make files. install.

xOchoa, 2020-11-09:
I'm getting the same issues, none of the media keys works after install

MatMoul, 2020-11-24:
No idea, tagging to need investigation

KiddingMe42, 2020-11-25:
Same issue with Ubuntu 18.04 (g213) - also the lighting is turned off after restart and can only enabled after plug the usb in again
xev nor showkeys -k can detect the media keys after make install
I removed the tool with make uninstall - reboot - and now it is working again.

saracalihan, 2021-03-03:
Same issue. OS: `5.10.18-1-Manjaro`, keyboard: `Logitech G G213 Prodigy`

saracalihan, 2021-03-03:
> Same issue. `5.10.18-1-Manjaro`, `Logitech G213 Prodigy`

I uninstall the project then install `g810-led-lib` from `aur` then the media keys works.

DarioMarzella, 2021-03-25:
Same issue here, Ubuntu 20.04.02, Keyboard G910.

AMDBartek, 2021-04-04:
Same issue on Kubuntu 20.10 with a Logitech G810 keyboard.

zocker-160, 2021-09-21:
the issue is when g810-led is using the USB interface, because when using libusb, the media keys are connected to the same endpoint and when g810-led requests access to said endpoint, it will unload the kernel driver, which handles the media keys.

The solution is to use hidapi, which does not need exclusive access to endpoints like libusb does.

MPatzek, 2023-03-24:
Is there a way to install hidapi mode via apt-install on ubuntu (mint) or is a manual installation needed?

zocker-160, 2023-03-24:
> Is there a way to install hidapi mode via apt-install on ubuntu (mint) or is a manual installation needed?

Hidapi mode needs to be supported by g810-led, installing additional packages will not fix the issue.

zocker-160, 2023-03-24:
Ah ok now I understand what you mean, sorry.

IIRC the Ubuntu repository has only one package available. If that is using libusb, then you will need to manually compile it.