MatMoul/g810-led #249: Frequent connect / disconnects from systemd-udevd

hnapel, 2020-10-12:
Upgraded to ubuntu 20.04, noted that the systemd-udevd was using close to 100% cpu,
then checked with udevadm monitor and noted high frequency of add/ bind / remove /unbind events for the
G513 keyboard like this:
<snip>
KERNEL[105975.409961] add      /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1/0003:046D:C33C.48F5/hidraw/hidraw7 (hidraw)
KERNEL[105975.409985] bind     /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1/0003:046D:C33C.48F5 (hid)
KERNEL[105975.410010] bind     /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1 (usb)
UDEV  [105975.410404] add      /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1/0003:046D:C33C.48E4/input/input74330/event11 (input)
UDEV  [105975.411549] bind     /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1/0003:046D:C33C.48E4 (hid)
UDEV  [105975.412858] bind     /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1 (usb)
UDEV  [105975.414204] remove   /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1/0003:046D:C33C.48E4/hidraw/hidraw7 (hidraw)
UDEV  [105975.414450] remove   /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1/0003:046D:C33C.48E4/input/input74330/event11 (input)
UDEV  [105975.414480] remove   /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1/usbmisc/hiddev2 (usbmisc)
UDEV  [105975.414526] remove   /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1/0003:046D:C33C.48E4/input/input74329/event10 (input)
UDEV  [105975.415432] remove   /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1/0003:046D:C33C.48E4/input/input74329 (input)
UDEV  [105975.415827] remove   /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1/0003:046D:C33C.48E4/input/input74330 (input)
UDEV  [105975.416728] unbind   /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1/0003:046D:C33C.48E4 (hid)
UDEV  [105975.417390] remove   /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1/0003:046D:C33C.48E4 (hid)
UDEV  [105975.418144] unbind   /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1 (usb)
KERNEL[105975.497801] remove   /devices/pci0000:00/0000:00:1c.4/0000:05:00.0/usb5/5-1/5-1.4/5-1.4.1/5-1.4.1:1.1/0003:046D:C33C.48F5/input/input74397/event10 (input)
<snip>

This goes on and on with a high rate. After running sudo make uninstall it went away, so I'm quite convinced this program is the issue. Please note I re-installed on the ubuntu 20.0.4 to pull in any updated libs, but the problem remains.

MatMoul, 2020-11-06:
How do you have installed g810-led?
On my system I use this file "/etc/udev/rules.d/g810-led.rules"
And I don't use systemd.
What services do you use for g810-led?

helix26j, 2021-01-19:
I was experiencing this same issue on my g910 and not having the RUN section at the end of the line in /lib/udev/rules.d/60-g810-led.rules appears to stop it and   This was occurring both on system boot but also when using a usb switch.  After removing the RUN the keyboard works immediately without having to wait for the disconnect/reconnects to stop.

My profile still applies on system boot (through systemd?).  Perhaps its some timing or conflict when the USB device is initializing?

helix26j, 2021-01-19:
Cannot seem to edit my previous comment.  I am using Linux Mint that is based on Ubuntu 20.04.