MatMoul/g810-led #253: Add tests to project

RoggeHaj, 2020-11-05:
As part of playing around with some code refactoring, I realized the task would be too cumbersome unless I had some proper unit tests to lean on. The test framework of choice was [https://github.com/catchorg/Catch2](https://github.com/catchorg/Catch2) since it's lightweight and only requires pulling one header file to get started.
So far I've implemented some tests for `utils.cpp` and hooked up the `makefile` to build the tests.
[image]

Would this code be of interest to the project? If so, I can polish it a bit and send a proper PR.

MatMoul, 2020-11-06:
It seem complicated...
I to tag to "need votes" to see what the community thinks about it

RoggeHaj, 2020-11-07:
It's not that complicated really.
```
$ git diff develop --name-only
.gitignore
makefile
tests/test_main.cpp
tests/test_utils.cpp
```
Obviously the test suite is not complete at this point but that's just a matter of adding test_ files in the tests directory when the need arises.