MatMoul/g810-led #254: G413 but no matching device found

theGeekyLad, 2020-11-26:
Came across this nice project on AUR and really wanted to test it out on my Arch PC. Was curious to try out `-a {color}` option as the G413 only sports red by nature. Nonetheless, I was surprised to see that it responds that the device isn't compatible! Here are some details:

## System as `uname -a`
Arch Linux - `5.9.6-arch1-1`

## Command Run
`g413-led -a #ff0000`

## Expectation
It should just light up the keyboard red.

## Result
`Matching or compatible device not found !`

## Additional Details
I also tried running `g413-led --list-keyboards` which knows my keyboard and prints the following:
```
Device: Logitech - G413 Carbon Mechanical Gaming Keyboard
	Vendor ID: <hidden>
	Product ID: <hidden>
	Serial Number: <hidden>
```

Any help will be really appreciated. :smile:

pe-nielsen, 2020-12-10:
I had a similar issue with a g810 that I wanted in all blue on Ubuntu. My issue was that I was running
`g810-led -a 0000ff`

when I needed to run it as root
`sudo g810-led -a 0000ff`

Maybe that helps?

PiecePaperCode, 2020-12-14:
`sudo` fixes the problem

MatMoul, 2021-01-25:
@theGeekyLad Some news?

theGeekyLad, 2021-01-26:
Thanks for suggestion @pe-nielsen! 

@MatMoul I'll be getting my replacement board on Thursday so I'll give it a try and post here.

theGeekyLad, 2021-02-07:
@PiecePaperCode `sudo` most certainly does fix the issue on my G413. :heart: Thanks a bunch!

closed 2021-02-07 by theGeekyLad