MatMoul/g810-led #255: Command 'g815-led' not found

pearsonradu, 2021-01-07:
I'm a new user of *g810-led* and first off I want to say how much I appreciate this project!

I am running Ubuntu 20.04.1 LTS and have run `sudo apt install g810-led` to install the application. I can run `g810-led --help` and the command is recognized but if I run `g815-led --help` the command is not found.

Is there something beyond `sudo apt install g810-led` that I need to do in order to register the `g815` command?

HerrEurobeat, 2021-01-24:
No, you need to use the `sudo g810-led` command even if your keyboard isn't a g810. If your keyboard is supported then it will just work.

MatMoul, 2021-01-25:
@pearsonradu Check the release... 
I'm pretty sure that with a `apt install g810-led` you're on the 0.3.3 release.
To support G815, you need the 0.4.2 release available in the testing repo.

elhennig, 2021-02-15:
I guess this is only a missing symbolic link in the installation package. But the binary is called form the udev rules in 80-g810-led.rules
Manually creating the link as root will help as well es changing the udev rules to call the g810-led binary instead.

closed 2021-05-13 by pearsonradu