MatMoul/g810-led #256: c++ "undefined reference" compilation ubuntu 20.10 g++

titicplusplus, 2021-01-15:
Hi
(sorry for my spelling mistakes, I am French)

I would like to create a program who synchronises my music and my led's keyboard (g213) .
To do that, i use the c++.
To install this repositery, i read the install.md, and executed its command lines : 

`sudo apt-get install git g++ make libhidapi-dev # for hidapi`
`git clone https://github.com/MatMoul/g810-led.git`
`cd g810-led`
`make lib # for hidapi`
`sudo make install-lib to install the libg810-led library.`
`sudo make install-dev`

No problem for that.

Afterwards, i executed this code : 

	#include <g810-led/Keyboard.h>

	int main()
	{
		LedKeyboard led;

		LedKeyboard::Color cool;

		cool.red = 0;
		cool.green = 0;
		cool.blue = 255;

		led.setAllKeys(cool);
		return 0;
	}

And when i executed that : `g++ main.cpp -o fichier_executable`

I have that : 

	/usr/bin/ld : /tmp/ccnOpkBM.o : in the fonction « main » :
	main.cpp:(.text+0x74) : rindefinite reference to « LedKeyboard::setAllKeys(LedKeyboard::Color) »
	/usr/bin/ld : main.cpp:(.text+0x88) : indefinite reference to « LedKeyboard::~LedKeyboard() »
	/usr/bin/ld : main.cpp:(.text+0xb1) : indefinite reference to « LedKeyboard::~LedKeyboard() »
	collect2: error: ld returned 1 exit status

(for information, i am translate the complisation because he was in french)

thank you in advance for your answers

pearsonk, 2021-01-15:
It looks like perhaps you did not include the linking, try along the lines of:
g++ main.cpp -o fichier_executable -lg810-led

titicplusplus, 2021-01-16:
Thank you, it works with your command line !

If someone arrived in this issue, here is my code:

	#include <g810-led/Keyboard.h>

	int main()
	{
		LedKeyboard led;

		std::string serial;
		uint16_t vendorID = 0x46d; //just for my keyboard G213
		uint16_t productID = 0xc336; //just for my keyboard G213

		if (!led.open(vendorID, productID, serial))
		{
			std::cout << "Matching or compatible device not found !" << std::endl;
			return 2;
		}
			

		LedKeyboard::Color cool;

		cool.red = 255;
		cool.green = 255;
		cool.blue = 0;

		led.setAllKeys(cool);
		
		return 0;
	}

titicplusplus, 2021-01-16:
I don"t know how to put "solved" :)

MatMoul, 2021-01-25:
@titicplusplus Just close the issue...
@pearsonk Thanks to help me!

closed 2021-01-25 by MatMoul