MatMoul/g810-led #263: G512 Carbon:

pedrohgodoys, 2021-04-07:
Hi, MatMoul. When I try to change colours, I have the following output:

`g512-led -a ffffff 
Matching or compatible device not found !`

Foi some reason this same problem appears when I intentionally try to use other keyboard commands, such as `g810-led -a ffffff`. Can you help me with that?
Thank you a lot, and great with this code. :D

jan-klima, 2021-04-10:
I think you need to start with g810-led and not g512-led...

Also - it seems this project is dead... I was waiting months for an answer...

pedrohgodoys, 2021-04-11:
> I think you need to start with g810-led and not g512-led...
> 
> Also - it seems this project is dead... I was waiting months for an answer...

I also tried with 810-led, but the same thing happened. 
It seems to be a good project, sadly to hear that. You know something similar to it to manage these colours?

armellarcier, 2021-04-19:
Try with sudo

pedrohgodoys, 2021-04-22:
> Try with sudo

I uninstalled it and reinstalled it. I don't know why, but this time worked well. But I have a key (ABNT keyboard) in a different colour now. I saw a pull about it, but I don't know who to apply it...

daholzfeind, 2022-01-06:
I had the same issue. To install it again run
```terminal
systemctl unmask g810-led-reboot      # unmask the service
sudo rm /usr/bin/g810-led             # remove old binary 
sudo make install                     # install again or via apt or whatever :)