MatMoul/g810-led #266: Does not want to install!

iRealistiX, 2021-07-13:
Hello! Simply:

niketpathak, 2021-09-02:
@iRealistiX There exists another project https://github.com/SebiTimeWaster/G213Colors too. I tried it but couldn't get it to work as yet. Another option is https://github.com/libratbag/libratbag

iRealistiX, 2021-09-03:
> @iRealistiX This repository doesn't seem to be maintained anymore (no updates for almost a year and a half). Anyway, there exists another project https://github.com/SebiTimeWaster/G213Colors too. I tried it but couldn't get it to work as yet. A third option is https://github.com/libratbag/libratbag

Thanks for the info. I totally forgotten the issue lol.

My fix for this is to edit the makefile to remove the systemd-run part and edited my xinitrc file the path to the program w/ the -p option poiting to /etc/g810-led/profile

Might add an option for the makefile and do PR