MatMoul/g810-led #268: Does not support G213 Prodigy anymore

niketpathak, 2021-09-02:
**System**: Ubuntu 20.04
**Device**: Bus 001 Device 003: ID 046d:c336 Logitech, Inc. Gaming Keyboard G213 (`lsusb`)

**Scenario 1**: Installed using repo (`apt install g810-led`)
`g213-led -a 00ff00`  ===> Matching or compatible device not found ! 

**Scenario 2**: Cloned this repository and installed as per [install.md](https://github.com/MatMoul/g810-led/blob/master/INSTALL.md)
`g213-led -a 00ff00`  ===> Matching or compatible device not found ! 

Also tried to test the keyboard with
`g213-led -dv 046d -dp c336 -tuk 4 -a 000000`  ===> Matching or compatible device not found ! 

Tried variations of the above command, but always receive the same error message.

niketpathak, 2021-09-05:
**Update**: For those having the same problem as me, there is another package https://gitlab.com/CalcProgrammer1/OpenRGB that worked for me like a charm. Feel free to give it a try if you're having the same issue.

Other alternates are:

https://github.com/SebiTimeWaster/G213Colors (tried it but couldn't get it to work). A third option is https://github.com/libratbag/libratbag that needs testing.

timoschroeder1213, 2021-09-10:
You need to prefix the command with `sudo`. That way it works on my machine. See: https://github.com/MatMoul/g810-led/issues/248

niketpathak, 2021-09-11:
I uninstalled it, restarted my computer and reinstalled it, and it apparently started working. No idea how. I installed it via repos => `sudo apt install g810-led`.

@timoschroeder1213 I was not required to use sudo to set colors.  Using  `g213-led -fx color keys 00ff00` worked without any permission issues. 

Closing this issue given that it's no longer reproducible.

closed 2021-09-11 by niketpathak