MatMoul/g810-led #269: `gkm 1` does not work on G815

zocker-160, 2021-09-21:
When running `g810-led -gkm 1` the default F1-F5 mapping does not get disabled as expected.
The only thing that happens, is that the LED on the M1 key gets switched off.

zocker-160, 2022-06-29:
hmm project seems dead :(

closed 2022-06-29 by zocker-160

mfvianna, 2022-11-19:
> hmm project seems dead :(

Had the same issue, but it does work.  You have to set the keyboard to software mode first:
`g815-leds --on-board-mode software`
(the keyboard leds will go all off, but you can load the profile again)

you can also set the on-board-mode on the profile as well, so these lines will help:

> obm software
> gkm 1

Then you can play with the mn and mr leds...

zocker-160, 2022-11-19:
Thanks for this info, maybe this is useful for others.

I personally don't need / use this anymore since I wrote my own user space driver, which allows me to map the G-keys as well. :)

mfvianna, 2022-11-19:
Hey @zocker-160 , could you please point me to your project? (if you have it publicly available, of course)

do you have any reference I could look for on getting G-key and M-key events in C?  I've a good experience on programming with libusb, but g815 doesn´t seem to produce any event in any of its /dev/input/event* device files.  I have no experience programming hdiapi, though.

A minimal documentation or code example on something similar to evtest (which btw I wrote my own before knowing about its existence) but using hidapi would be perfect.

zocker-160, 2022-11-19:
@mfvianna sure https://github.com/zocker-160/keyboard-center

> g815 doesn´t seem to produce any event in any of its /dev/input/event* device files

Yeah that is true for all Logitech Keyboards and the reason why I created the project above.
I take the super lazy approach of just listening to libhidraw data coming from the keyboard and then I just use a simple data table to compare it to known commands (usually only Gkeys, Mkeys, volume wheel and media buttons).

G815 is also supported thanks to a community member contributing.

mfvianna, 2022-11-20:
Hey, thank you @zocker-160 , it seems libhidraw is simple enough =)

jfi, actually my ooooold Logitech G15 ( https://techgage.com/article/logitech_g15_revision_2/ ) did generate events for G and M keys, in a separate /dev/input/event* (like G815 does, but only for the media keys).  I had a similar project as yours, probably over a decade ago... :-)

Will explore libhidraw a bit.  I intend to develop a similar project again, aiming for visual reality (like using real keyboard images, much like GHUB), hopefully we can share some knowledge and I can return the favor and kindness of quickly responding my msgs.

Thank you my friend.

zocker-160, 2022-11-20:
@mfvianna libhidraw is not hard to use, you can check my Python bindings in the repo, it should help you to get started if you are stuck.