MatMoul/g810-led #273: LED coloring through KVM switch

jonathanpullano, 2021-11-04:
I realize this is a bit niche, but I thought I would ask anyway.

I have a Logitech G910, and am running Ubuntu 20.04.3 LTS, connected to my pc via a TESmart Dual HDMI 4x2 Dual Monitor KVM Switch (https://www.amazon.com/TESmart-Monitor-Switch-Updated-Support/dp/B083NGXHWG/ref=sr_1_6?crid=2SMSKDSUI4LKI&keywords=dual+monitor+kvm&qid=1636051113&sprefix=dual+monitor+kvm%2Caps%2C211&sr=8-6)

Your tool appears to install correctly, as I can use it if I directly connect the G910 to my PC, and the color updates as expected via the -a command. But when I connect it via the KVM the color pattern is reset to rainbow, and running -a has no effect. 

When I use the command
g910-led --list-keyboards
I see a vendor ID and Product ID (but no serial number).

Any thoughts? Thanks

snoopdouglas, 2021-11-10:
I realise this is a slightly different setup, but I'm currently running PopOS 20.04 and my G513 is working without issue through a KVM switch (AV Access 4KSW41-KVM).

rienafairefr, 2023-09-03:
Not an issue with the library, this is more like a +1. Slighly different setup but same KVM manufacturer, I get the same result (uncontrollable led coloring on the keyboard and no serial number) for me, KVM TESmart HKS0201B2U