MatMoul/g810-led #274: Matching or compatible device not found

buxx, 2021-11-10:
Hello, I just installed this program from debian packages. Program can find my keyboard when list, but fail when apply a color :

̀```
➜  Projets g413-led --list-keyboards
Device: Logitech - G413 Carbon Mechanical Gaming Keyboard
	Vendor ID: 046d
	Product ID: c33a
	Serial Number: 1671306C3832
➜  Projets g413-led -a 00ff00       
Matching or compatible device not found !
```

What should I do ?

robtech21, 2021-11-12:
This happened to me with my G213 and I was able to fix it quickly. In your case you'd probably want to do:

`g413-led -ds 1671306C3832`

and then try setting a theme

elidrissidev, 2021-11-12:
Had the same error using Manjaro KDE, running the command `sudo` did the trick for me. Hope it helps someone.

robtech21, 2021-11-17:
> Had the same error using Manjaro KDE, running the command with `sudo` did the trick for me. Hope it helps someone.

Yes, using sudo also helps.

buxx, 2021-11-18:
Thanks ! I close the issue.

closed 2021-11-18 by buxx