MatMoul/g810-led #281: g213 Prodigy and Mint Mate

VladBg88, 2022-01-22:
Hi folks, 

I am completely new to Linux and I love the OS, but I do not have so much coding experience. Can you please explain to me as for complete noob which commands I can use to enable my RGB LEDs on Logitech G213? 

I am using Linux Mint Mate. 

I have installed the package of libg810-led0 0.3.9- 1build1. 

Thanks in advance!

VladBg88, 2022-01-22:
My experience so far is: 

vladislav@vladislav-System-Product-Name:~$ apt install g810-led
Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for ^Citing for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 870877 (synaptic)... 56sb/dpkg/lock-frontend. It is heWaiting for cvladislav@vladislav-System-Product-Name:~$ ^Cock-frontend. It is held by process 870877 (synaptic)... 47s /var/lib/dpkg/lock-frontend. It is held by process 8708vladislav@vladislav-System-Product-Name:~$ tic)... 43st lock /var/lib/dpkg/lock-frontend. It is held by process 870877 (synaptic)... 41ss held Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 870877 (synaptic)... 40sot get lock /var/lib/dpkg/lock-frontend. It is held by process 870877 (synaptic)... 38sfrontend. It is held by process 870877 (synaptic)... 36saptic)... 34s

elidrissidev, 2022-01-23:
Hey, I have the same keyboard and I'm currently using Manjaro but I used to use a debian based distro. The error you're facing has to do with the package manager being in-use by another process, all you need to do is wait a little and try again. After following the installation instructions, you'll need to run the command the first time with `sudo` e.g: `sudo g213-led -a 00ff00`

VladBg88, 2022-01-23:
Hi Moahamed Elidrissi,
thank you for your email!

I am new to Linux and I am testing now on my laptop Manjaro too.

Greetings,
Vladislav