MatMoul/g810-led #285: error: packages failed to build: g810-led-git-0.4.2-2

Siiire, 2022-04-03:
Getting this error on arch, been a few days now. Is there anything I can do to help, like paste certain logs/system info? I have no knowledge when it comes to software development.

Shyrak, 2022-04-04:
Another Arch user here. Seems to be a typo in the git URL on the PKGBUILD.
Using `yay` you can edit it before installing with the following command (I'm sure other AUR managers will have something similar):
```bash
yay --editmenu -S g810-led-git 
```
Once you get the editor up, line 20:
```
source=("git+https://github.com/MatMoul/${_appname}.git") 
```

Should be like this:
```
source=("git://github.com/MatMoul/${_appname}.git")
```

Hope this helps!

Siiire, 2022-04-05:
> Another Arch user here. Seems to be a typo in the git URL on the PKGBUILD. Using `yay` you can edit it before installing with the following command (I'm sure other AUR managers will have something similar):
> 
> ```shell
> yay --editmenu -S g810-led-git 
> ```
> 
> Once you get the editor up, line 20:
> 
> ```
> source=("git+https://github.com/MatMoul/${_appname}.git") 
> ```
> 
> Should be like this:
> 
> ```
> source=("git://github.com/MatMoul/${_appname}.git")
> ```
> 
> Hope this helps!

Thank you for your reply! I tried your solution, replacing yay with paru, it didn't work. I tried to look for a similar command using paru -h but I couldn't find anything.

Shyrak, 2022-04-07:
I'm not familiar with paru, but seems to offer package review as a default. 
When installing the package, a prompt to "Proceed to review?" should appear (at least from I can gather from the video at their github). There you might be able to edit the PKGBUILD file and try the changes I suggested.
I'm sorry I can't be of much help at this point, but I've never used paru, so I can't give you a more detailed explanation.

Other alternative would be to go to the manual route described [here](https://wiki.archlinux.org/title/Arch_User_Repository#Installing_and_upgrading_packages).

closed 2022-04-10 by Siiire