MatMoul/g810-led #289: [G815] Alias "num." is working but "numdot" and "num_dot" are not

ercksen, 2022-05-26:
I found that only the "num." alias is working to change the color of the period button of my numpad. The other proposed ones "numdot" and "num_dot" are not working.

This probably also affects other devices.

Version: 0.4.2

val-int1, 2022-06-04:
Can confirm with a G513 keyboard. When running the following command
`g513-led -g numeric 000000`
the whole numpad (including the dot key) turns off, but while running these commands
`g513-led -k numdot ff0000`
`g513-led -k num_dot 00ff00`
`g513-led -k num. 0000ff`
the numpad dot key stays off except with the `num.` alias.

I haven't found any other key where this problem happens.

```
g810-led version: 0.4.2 (installed via apt-get)
OS: Pop!_OS 21.10
DE: KDE Plasma 5.22.5
```

https://user-images.githubusercontent.com/25990664/171971229-254147a5-8ba9-4cad-80b2-82b4f2ede9e0.mp4

val-int1, 2022-06-09:
I figured out why it doesn't work. [This line](https://github.com/MatMoul/g810-led/blob/master/src/helpers/utils.cpp#L178) checks for `num_period` and `numperiod`, even though [--help-keys](https://github.com/val-int1/g810-led/blob/master/src/helpers/help.cpp#L226) specifies `num_dot` and `numdot`. Using the former keywords works for me, I've also made [a fork](https://github.com/val-int1/g810-led) that allows using all those keywords.