MatMoul/g810-led #290: G915 : any support ?

Esysteme, 2022-08-25:
Hello,
thx for your work.

What do you need to add g915 there ?

Esysteme, 2022-08-25:
```
g810-led -dv 046d -dp c33e -tuk 1 -a 000000
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
root@aurelien-xps:/srv/code/g810-led# g810-led -dv 046d -dp c33e -tuk 2 -a 000000
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
root@aurelien-xps:/srv/code/g810-led# g810-led -dv 046d -dp c33e -tuk 3 -a 000000
Error: Can not write to hidraw, try with the libusb version
root@aurelien-xps:/srv/code/g810-led# g810-led -dv 046d -dp c33e -tuk 4 -a 000000
Matching or compatible device not found !

Esysteme, 2022-08-25:
```
root@aurelien-xps:/srv/code/g810-led# g810-led -dv 046d -dp c541 -tuk 4 -a 000000
Matching or compatible device not found !
root@aurelien-xps:/srv/code/g810-led# g810-led -dv 046d -dp c541 -tuk 1 -a 000000
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
root@aurelien-xps:/srv/code/g810-led# g810-led -dv 046d -dp c541 -tuk 2 -a 000000
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
Error: Can not write to hidraw, try with the libusb version
root@aurelien-xps:/srv/code/g810-led# g810-led -dv 046d -dp c541 -tuk 3 -a 000000
Error: Can not write to hidraw, try with the libusb version

jason-pollock, 2022-09-04:
@Esysteme ever get this working? I'm wondering the same thing. Would love to see support for the 915.

Esysteme, 2022-09-07:
https://womanonrails.com/logitech-g915-tkl

there is this but it's wasnt working for me

ausinch, 2023-02-01:
I am using **ratbagctl** until G915 is added to g810-led.
ratbagctl can't do individual keys but can set the whole kb colour as well as a G502 mouse.
I wrote a command (in python) based on ratbagctl for my son to change and sync his mouse & kb colours.
I will post the script to github when I get time.