MatMoul/g810-led #303: g810-led does not compile with gcc13.1.1 using AUR pkgbuild on Archlinux

dasnoopy, 2023-05-07:
as per title, on archlinux using the [AUR](https://aur.archlinux.org/packages/g810-led-git) pkgbuild the app does not compile..

see also this recent [comment](https://aur.archlinux.org/packages/g810-led-git#comment-913466) in the AUR page...

```
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --enable-languages=ada,c,c++,d,fortran,go,lto,objc,obj-c++ --enable-bootstrap --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --with-build-config=bootstrap-lto --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-libstdcxx-backtrace --enable-link-serialization=1 --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-werror
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 13.1.1 20230429 (GCC) 
```

```
=> Making package: g810-led-git 0.4.3.r1.efa3c35-1 (dom 7 mag 2023, 20:51:18)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating g810-led git repo...
==> Validating source files with sha256sums...
    g810-led ... Skipped
==> Extracting sources...
  -> Creating working copy of g810-led git repo...
Reset branch 'makepkg'
==> Removing existing $pkgdir/ directory...
==> Starting build()...
g++ -Dhidapi -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection -Wp,-D_GLIBCXX_ASSERTIONS -std=gnu++11 -DVERSION=\"0.4.3\" -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now src/main.cpp src/helpers/help.cpp src/helpers/utils.cpp src/classes/Keyboard.cpp -o bin/g810-led -lhidapi-hidraw
In file included from src/main.cpp:23:
src/helpers/help.h:24:14: warning: elaborated-type-specifier for a scoped enum must not use the ‘class’ keyword
   24 |         enum class KeyboardFeatures : uint16_t {
      |         ~~~~ ^~~~~
      |              -----
src/helpers/help.h:24:37: error: found ‘:’ in nested-name-specifier, expected ‘::’
   24 |         enum class KeyboardFeatures : uint16_t {
      |                                     ^
      |                                     ::
src/helpers/help.h:24:20: error: ‘KeyboardFeatures’ has not been declared
   24 |         enum class KeyboardFeatures : uint16_t {
      |                    ^~~~~~~~~~~~~~~~
src/helpers/help.h:24:48: error: expected unqualified-id before ‘{’ token
   24 |         enum class KeyboardFeatures : uint16_t {
      |                                                ^
src/helpers/help.h:59:16: error: ‘KeyboardFeatures’ does not name a type
   59 |         inline KeyboardFeatures operator|(KeyboardFeatures a, KeyboardFeatures b);
      |                ^~~~~~~~~~~~~~~~
src/helpers/help.h:61:9: error: ‘KeyboardFeatures’ does not name a type
   61 |         KeyboardFeatures getKeyboardFeatures(std::string cmdName);
      |         ^~~~~~~~~~~~~~~~
In file included from src/helpers/help.cpp:17:
src/helpers/help.h:24:14: warning: elaborated-type-specifier for a scoped enum must not use the ‘class’ keyword
   24 |         enum class KeyboardFeatures : uint16_t {
      |         ~~~~ ^~~~~
      |              -----
src/helpers/help.h:24:37: error: found ‘:’ in nested-name-specifier, expected ‘::’
   24 |         enum class KeyboardFeatures : uint16_t {
      |                                     ^
      |                                     ::
src/helpers/help.h:24:20: error: ‘KeyboardFeatures’ has not been declared
   24 |         enum class KeyboardFeatures : uint16_t {
      |                    ^~~~~~~~~~~~~~~~
src/helpers/help.h:24:48: error: expected unqualified-id before ‘{’ token
   24 |         enum class KeyboardFeatures : uint16_t {
      |                                                ^
src/helpers/help.h:59:16: error: ‘KeyboardFeatures’ does not name a type
   59 |         inline KeyboardFeatures operator|(KeyboardFeatures a, KeyboardFeatures b);
      |                ^~~~~~~~~~~~~~~~
src/helpers/help.h:61:9: error: ‘KeyboardFeatures’ does not name a type
[... 249 lines cut ...]
src/helpers/help.cpp:339:40: error: ‘KeyboardFeatures’ is not a class, namespace, or enumeration
  339 |                         if((features | KeyboardFeatures::setall) == features)
      |                                        ^~~~~~~~~~~~~~~~
src/helpers/help.cpp:341:40: error: ‘KeyboardFeatures’ is not a class, namespace, or enumeration
  341 |                         if((features | KeyboardFeatures::setgroup) == features)
      |                                        ^~~~~~~~~~~~~~~~
src/helpers/help.cpp:343:40: error: ‘KeyboardFeatures’ is not a class, namespace, or enumeration
  343 |                         if((features | KeyboardFeatures::setkey) == features) {
      |                                        ^~~~~~~~~~~~~~~~
src/helpers/help.cpp:353:21: error: ‘features’ was not declared in this scope
  353 |                 if((features | KeyboardFeatures::intensity) == features) {
      |                     ^~~~~~~~
src/helpers/help.cpp:353:32: error: ‘KeyboardFeatures’ is not a class, namespace, or enumeration
  353 |                 if((features | KeyboardFeatures::intensity) == features) {
      |                                ^~~~~~~~~~~~~~~~
src/helpers/help.cpp:360:21: error: ‘features’ was not declared in this scope
  360 |                 if((features | KeyboardFeatures::setkey) == features) {
      |                     ^~~~~~~~
src/helpers/help.cpp:360:32: error: ‘KeyboardFeatures’ is not a class, namespace, or enumeration
  360 |                 if((features | KeyboardFeatures::setkey) == features) {
      |                                ^~~~~~~~~~~~~~~~
make: *** [makefile:35: bin/g810-led] Error 1
==> ERROR: A failure occurred in build().
    Aborting...
```

Gy0m, 2023-05-26:
Same error here

XaserLE, 2023-05-31:
I got the same error. There seems to be a missing `#include <cstdint>` in `src/helpers/help.h`. After adding it in the downloaded source the `make` command runs fine. When using `makepkg` on Arch Linux, the `-e` switch uses the modified sources without redownloading. So the solution for me was:

1. Download from AUR
2. Modifiy `g810-led/src/g810-led-0.4.3/src/helpers/help.h` by adding `#include <cstdint>` (e.g. in front of the `#include <iostream>`)
3. `$ makepkg -e`
4. `$ pacman -U *.zst`

LRitzdorf, 2023-06-01:
Would be closed by #302 (for cross-ref purposes)

abarocio80, 2023-08-12:
I have successfully installed this software in archlinux either by modifying a line in `src/helpers/help.h` like this

```
- 	enum class KeyboardFeatures : uint16_t {
+	enum KeyboardFeatures {
```
Or (in case that is needed and works in other systems) by modifying that line in the build process from the PKGBUILD script like this:

```
build() {
	cd "${_githubrepo}-${pkgver}"
+        sed 's/enum class KeyboardFeatures : uint16_t {/enum KeyboardFeatures {/' src/helpers/help.h
	make bin
}
```

Which way is less conflictive?