MatMoul/g810-led #304: G413 TLK SE - 046d:c34a

MightyPork, 2023-05-10:
My keyboard isn't detected. The PID is c34a instead of c33a.

Running this on Arch linux, compiled with g++-11 (I edited the PKGBUILD like so: `make LIB=libusb CXX=g++-11 bin`).

I found and edited the VID:PID list in the header file `Keyboard.h`, but that didn't help, now it segfaults when trying to list keyboards.

```
Thread 1 "g413-led" received signal SIGSEGV, Segmentation fault.
libusb_close (dev_handle=0xffffffffffffff98) at /usr/src/debug/libusb/libusb-1.0.26/libusb/core.c:1452
Downloading source file /usr/src/debug/libusb/libusb-1.0.26/libusb/core.c
1452    void API_EXPORTED libusb_close(libusb_device_handle *dev_handle)   
```

gdb doesn't say much...

it's probably something I'm doing wrong, I had similar problem with keyleds

LRitzdorf, 2023-06-01:
I seem to be encountering a similar issue with my G413 TKL SE, which also has a product ID of `c34a`.

I've built from the AUR (and incorporating #302), but upon running `g413-led -dv 046d -dp c34a -tuk 1 -a 000000`, I receive a series of `Can not write to hidraw` errors (both with and without sudo).

I'm not particularly familiar with device interfaces on Linux, but would be happy to provide further details or test results if desired!

MightyPork, 2023-06-02:
I wasted so much time on this, installing windows in a VM, installing various DLLs from old windows to make the horrible app run on win 11, and eventually had to conclude the particular model is a striped down version that does not support anything from the protocol - it can't even remember the last brightness level.

The Logitech hub app detects it and just says you don't need it to use the keyboard. You can't do anything else. So useful.

Had I known this, I wouldn't buy it.

So this issue is probably baseless and can be closed. You have to get used to the silly animation at boot, then it's generally fine on the default settings (max brightness)

LRitzdorf, 2023-06-02:
Yeah, I've determined the same. Thankfully, my keyboard is still within the return period.

It might be worth noting in the README that the SE version is not supported, just to be explicit about its lack of hardware capability?

closed 2023-06-02 by MightyPork

antoinefrancispierre, 2023-07-19:
Hello,

I have a G413 TLK SE as well and cannot make the tool working with it. I tried running `g413-led -dv 046d -dp c34a -tuk [1-4] -a 000000` but keep getting `Error: Can not write to hidraw, try with the libusb version`.

Can you please re-open the case? Anyone has a hint where to look at to fix this for this specific keyboard?

Thanks for your help!

LRitzdorf, 2023-07-19:
Based on the OP's detailed investigation and my experience, it seems that the SE variant simply does not support being controlled from the attached computer. Blame Logitech, not this project.