MatMoul/g810-led #311: Build failed

hamburgerghini1, 2024-05-10:
`[tommi@archlinux g810-led]$ make bin
g++ -Dhidapi -Wall -O2 -std=gnu++11 -DVERSION=\"0.4.3\"  src/main.cpp src/helpers/help.cpp src/helpers/utils.cpp src/classes/Keyboard.cpp -o bin/g810-led -lhidapi-hidraw
In file included from src/main.cpp:23:
src/helpers/help.h:24:14: warning: elaborated-type-specifier for a scoped enum must not use the ‘class’ keyword
   24 |         enum class KeyboardFeatures : uint16_t {
      |         ~~~~ ^~~~~
      |              -----
src/helpers/help.h:24:37: error: found ‘:’ in nested-name-specifier, expected ‘::’
   24 |         enum class KeyboardFeatures : uint16_t {
      |                                     ^
      |                                     ::
src/helpers/help.h:24:20: error: ‘KeyboardFeatures’ has not been declared
   24 |         enum class KeyboardFeatures : uint16_t {
      |                    ^~~~~~~~~~~~~~~~
src/helpers/help.h:24:48: error: expected unqualified-id before ‘{’ token
   24 |         enum class KeyboardFeatures : uint16_t {
      |                                                ^
src/helpers/help.h:59:16: error: ‘KeyboardFeatures’ does not name a type
   59 |         inline KeyboardFeatures operator|(KeyboardFeatures a, KeyboardFeatures b);
      |                ^~~~~~~~~~~~~~~~
src/helpers/help.h:61:9: error: ‘KeyboardFeatures’ does not name a type
   61 |         KeyboardFeatures getKeyboardFeatures(std::string cmdName);
      |         ^~~~~~~~~~~~~~~~
In file included from src/helpers/help.cpp:17:
src/helpers/help.h:24:14: warning: elaborated-type-specifier for a scoped enum must not use the ‘class’ keyword
   24 |         enum class KeyboardFeatures : uint16_t {
      |         ~~~~ ^~~~~
      |              -----
src/helpers/help.h:24:37: error: found ‘:’ in nested-name-specifier, expected ‘::’
   24 |         enum class KeyboardFeatures : uint16_t {
      |                                     ^
      |                                     ::
src/helpers/help.h:24:20: error: ‘KeyboardFeatures’ has not been declared
   24 |         enum class KeyboardFeatures : uint16_t {
      |                    ^~~~~~~~~~~~~~~~
src/helpers/help.h:24:48: error: expected unqualified-id before ‘{’ token
   24 |         enum class KeyboardFeatures : uint16_t {
      |                                                ^
src/helpers/help.h:59:16: error: ‘KeyboardFeatures’ does not name a type
   59 |         inline KeyboardFeatures operator|(KeyboardFeatures a, KeyboardFeatures b);
      |                ^~~~~~~~~~~~~~~~
src/helpers/help.h:61:9: error: ‘KeyboardFeatures’ does not name a type
   61 |         KeyboardFeatures getKeyboardFeatures(std::string cmdName);
      |         ^~~~~~~~~~~~~~~~
src/helpers/help.cpp:27:16: error: ‘KeyboardFeatures’ does not name a type
   27 |         inline KeyboardFeatures operator|(KeyboardFeatures a, KeyboardFeatures b) {
      |                ^~~~~~~~~~~~~~~~
src/helpers/help.cpp:31:9: error: ‘KeyboardFeatures’ does not name a type
   31 |         KeyboardFeatures getKeyboardFeatures(string cmdName) {
      |         ^~~~~~~~~~~~~~~~
src/helpers/help.cpp: In function ‘void help::usage(char*)’:
src/helpers/help.cpp:48:17: error: ‘KeyboardFeatures’ was not declared in this scope
   48 |                 KeyboardFeatures features = getKeyboardFeatures(cmdName);
      |                 ^~~~~~~~~~~~~~~~
src/helpers/help.cpp:55:21: error: ‘features’ was not declared in this scope
   55 |                 if((features | KeyboardFeatures::setall) == features)
      |                     ^~~~~~~~
src/helpers/help.cpp:55:32: error: ‘KeyboardFeatures’ is not a class, namespace, or enumeration
   55 |                 if((features | KeyboardFeatures::setall) == features)
      |                                ^~~~~~~~~~~~~~~~
src/helpers/help.cpp:57:21: error: ‘features’ was not declared in this scope
   57 |                 if((features | KeyboardFeatures::setgroup) == features)
      |                     ^~~~~~~~
src/helpers/help.cpp:57:32: error: ‘KeyboardFeatures’ is not a class, namespace, or enumeration
   57 |                 if((features | KeyboardFeatures::setgroup) == features)
      |                                ^~~~~~~~~~~~~~~~
src/helpers/help.cpp:59:21: error: ‘features’ was not declared in this scope
   59 |                 if((features | KeyboardFeatures::setkey) == features)
      |                     ^~~~~~~~
src/helpers/help.cpp:59:32: error: ‘KeyboardFeatures’ is not a class, namespace, or enumeration
[... 218 lines cut ...]
src/helpers/help.cpp:337:32: error: ‘KeyboardFeatures’ is not a class, namespace, or enumeration
  337 |                 if((features | KeyboardFeatures::commit) == features) {
      |                                ^~~~~~~~~~~~~~~~
src/helpers/help.cpp:339:40: error: ‘KeyboardFeatures’ is not a class, namespace, or enumeration
  339 |                         if((features | KeyboardFeatures::setall) == features)
      |                                        ^~~~~~~~~~~~~~~~
src/helpers/help.cpp:341:40: error: ‘KeyboardFeatures’ is not a class, namespace, or enumeration
  341 |                         if((features | KeyboardFeatures::setgroup) == features)
      |                                        ^~~~~~~~~~~~~~~~
src/helpers/help.cpp:343:40: error: ‘KeyboardFeatures’ is not a class, namespace, or enumeration
  343 |                         if((features | KeyboardFeatures::setkey) == features) {
      |                                        ^~~~~~~~~~~~~~~~
src/helpers/help.cpp:353:21: error: ‘features’ was not declared in this scope
  353 |                 if((features | KeyboardFeatures::intensity) == features) {
      |                     ^~~~~~~~
src/helpers/help.cpp:353:32: error: ‘KeyboardFeatures’ is not a class, namespace, or enumeration
  353 |                 if((features | KeyboardFeatures::intensity) == features) {
      |                                ^~~~~~~~~~~~~~~~
src/helpers/help.cpp:360:21: error: ‘features’ was not declared in this scope
  360 |                 if((features | KeyboardFeatures::setkey) == features) {
      |                     ^~~~~~~~
src/helpers/help.cpp:360:32: error: ‘KeyboardFeatures’ is not a class, namespace, or enumeration
  360 |                 if((features | KeyboardFeatures::setkey) == features) {
      |                                ^~~~~~~~~~~~~~~~
make: *** [makefile:35: bin/g810-led] Error 1`

julian18, 2024-06-07:
Apparently the real error is that uint16_t is not defined.
I fixed this by adding `#include <inttypes.h>` to `src/helpers/help.h`
There is already a [pull request](https://github.com/MatMoul/g810-led/pull/305) that adds cstdint instead.

Vezxe, 2024-06-08:
> Apparently the real error is that uint16_t is not defined. I fixed this by adding `#include <inttypes.h>` to `src/helpers/help.h` There is already a [pull request](https://github.com/MatMoul/g810-led/pull/305) that adds cstdint instead.

Had the same issue, adding this fixed it